nordicsemi/Kotlin-BLE-Library #21: Question : whyuse suspendCoroutine instead of suspendCancellableCoroutine

corentin-c, 2023-10-12:
Hello,

Thank you for this interesting library. I am using the Android BLE library, I tried to integrate this one to test it, and I wanted to discuss an issue before opening a PR. 

One of the issue I had with the previous library when trying to use Kotlin, is thas operations queued using suspend were sometimes never completing. So I tried wrapping the operations in a withTimeout, but it cannot work because the old library used suspendCoroutine instead of suspendCancellableCoroutine, which means that the withTimeout cannot cancel the operation and timeout correctly.

Here, I see that you are using the same (for exemple here https://github.com/NordicSemiconductor/Kotlin-BLE-Library/blob/3a96177b43d5d35af8f97f9d439f8f374c6c8f42/client/src/main/java/no/nordicsemi/android/kotlin/ble/client/main/service/ClientBleGattCharacteristic.kt#L198), and I'm wondering if there is a particular reason why you don't use suspendCancellableCoroutine instead of suspendCoroutine ? And if not, could it be used to allow for timeouts to work ?

sylwester-zielinski, 2023-10-13:
Hi,

Thank you for your feedback. Let me explain the situation. In short words we use `suspendCoroutine` to change JAVA callbacks into suspend functions. First a `BluetoothGatt's` method is invoked, then we suspend a function, later we get a callback in `BluetoothGattCallback` and we resume the function.
If we do 2 consecutive calls at the same time then Android will ignore the latter and the suspend function will never resume.
That's why invoking consecutive calls to `BluetoothGatt` are protected by a `Mutex`.

If we make these functions cancellable then I can imagine a situation when we release the mutex before getting a callback from `BluetoothGattCallback`.
Then making another call to `BluetoothGatt` will be ignored.

I can see your point and I think those functions should be cancellable too, but to do so I guess we will have to move `mutex.unlock()` invocation from `ClientBleGattCharacteristic` to `ClientBleGattCallback` to be sure that those will be invoked even if the coroutine has been cancelled and make sure that next call to `BluetoothGatt` will not be invoked until callback returns.

corentin-c, 2023-10-13:
Would a continuation.invokeOnCancellation not be enough ? Like this :

 

```kotlin
    suspend fun write(value: DataByteArray, writeType: BleWriteType = BleWriteType.DEFAULT) {
        mutex.lock()
        val stacktrace = Exception() //Helper exception to display valid stacktrace.
        return suspendCancellableCoroutine { continuation ->
            logger.log(Log.DEBUG, "Write to characteristic - start, uuid: $uuid, value: $value, type: $writeType")
            validateWriteProperties(writeType)
            pendingWriteEvent = {
                pendingWriteEvent = null
                if (it.status.isSuccess) {
                    logger.log(Log.INFO, "Value written: $value to $uuid")
                    continuation.resume(Unit)
                } else {
                    logger.log(Log.ERROR, "Write to characteristic - error, uuid: $uuid, result: ${it.status}")
                    continuation.resumeWithException(GattOperationException(it.status, cause = stacktrace))
                }
                mutex.unlock()
            }
           // HERE
            continuation.invokeOnCancellation {
                mutex.unlock()
            }
            gatt.writeCharacteristic(characteristic, value, writeType)
        }
    }
```

 

This way we make sure the mutex is unlocked when the coroutine is cancelled, we still have the mutex here so next call to BluetoothGatt will not be invoked, and we don't have to move the mutex to  ClientBleGattCallback . Or am I missing something ?

sylwester-zielinski, 2023-10-17:
In my opinion it works this way: when you call BluetoothGatt then you must wait for callback otherwise your next request will not get its callback invoked e.g. If you call BluetoothGatt first time, then call BluetoothGatt second time then you will get callback for the first call, but the callback for the second call will never be invoked. Maybe this was your use case? That's why there is a mutex, to synchronise this requests one by one. 

It's little surprising to me that the callback may never be called if you follow this rule. If this is not the case, then we can do as you suggested but I can see one potential problem with the solution: if we release mutex onTimeout too soon, when callback is still processing, then we will release the mutex. As a result another write to characteristic will be ignored by Android and we will end up with a function which is suspended forever (suspended not on Mutex, but on waiting for pendingWriteEvent to be invoked).

corentin-c, 2023-10-17:
I see your point, I am using the old BLE library without mutex on my end because I thought the library was managing it on its own, maybe that's why we encontered this issue in the first place.
We should not encounter it with this  new library, if it's the case I will reopen this.
Thank you for your answers

closed 2023-10-17 by corentin-c

philips77, 2023-10-18:
Just to say few word about the old BLE library. There are cancellable requests and non-cancellable ones. For example `ConnectRequest` or `Sleep` is cancellable, as you may abort it any time, but `WriteRequest` isn't due to what Sylwester has written - the callback is always expected - with success or failure.

But there are also nuances. For example, the `WriteRequest` has `.split()` method, which automatically splits long data using the `Spitter` (defaults to MTU-log packets). In that case, if the request is to be sent in several packets, the whole request should be cancellable. As you should be able to abort after sending each packet, that is after getting `onCharacteristicWrite` callback.

> One of the issue I had with the previous library when trying to use Kotlin, is thas operations queued using suspend were sometimes never completing. So I tried wrapping the operations in a [...]

Did you open an issue to the old BLE library?

corentin-c, 2023-10-18:
No I did not, it was impossible to reproduce consistently for us, so we could not debug it. This might come from the fact that we didn't use any mutex on our end, I'm not sure

corentin-c, 2023-10-18:
Now that I think about it, I had opened an issue to make split write request cancellable, and because it is not for now, we still use
```kotlin
    override fun chunk(message: ByteArray, index: Int, maxLength: Int): ByteArray? {
        if (isCanceled) return null
        ``` 
to cancel them. Maybe that coule be causing this issue, I will check