nordicsemi/Kotlin-BLE-Library #39: Advertising fails to enable in the Blinky App

reactormonk, 2023-10-25:
When hitting the "Advertise" button, this shows up in the logs:

```
isLeEnabled(): ON
java.lang.IllegalArgumentException: Advertising data too big
	at android.bluetooth.le.BluetoothLeAdvertiser.startAdvertisingSet(BluetoothLeAdvertiser.java:408)
	at android.bluetooth.le.BluetoothLeAdvertiser.startAdvertisingSet(BluetoothLeAdvertiser.java:262)
	at no.nordicsemi.android.kotlin.ble.advertiser.BleAdvertiserOreo$advertise$1.invokeSuspend(BleAdvertiserOreo.kt:78)
	at no.nordicsemi.android.kotlin.ble.advertiser.BleAdvertiserOreo$advertise$1.invoke(Unknown Source:8)
	at no.nordicsemi.android.kotlin.ble.advertiser.BleAdvertiserOreo$advertise$1.invoke(Unknown Source:4)
	at kotlinx.coroutines.flow.ChannelFlowBuilder.collectTo$suspendImpl(Builders.kt:320)
	at kotlinx.coroutines.flow.ChannelFlowBuilder.collectTo(Unknown Source:0)
	at kotlinx.coroutines.flow.CallbackFlowBuilder.collectTo(Builders.kt:334)
	at kotlinx.coroutines.flow.internal.ChannelFlow$collectToFun$1.invokeSuspend(ChannelFlow.kt:60)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)
	at kotlinx.coroutines.EventLoop.processUnconfinedEvent(EventLoop.common.kt:68)
	at kotlinx.coroutines.internal.DispatchedContinuationKt.resumeCancellableWith(DispatchedContinuation.kt:375)
	at kotlinx.coroutines.intrinsics.CancellableKt.startCoroutineCancellable(Cancellable.kt:30)
	at kotlinx.coroutines.intrinsics.CancellableKt.startCoroutineCancellable$default(Cancellable.kt:25)
	at kotlinx.coroutines.CoroutineStart.invoke(CoroutineStart.kt:110)
	at kotlinx.coroutines.AbstractCoroutine.start(AbstractCoroutine.kt:126)
	at kotlinx.coroutines.BuildersKt__Builders_commonKt.launch(Builders.common.kt:56)
	at kotlinx.coroutines.BuildersKt.launch(Unknown Source:1)
	at kotlinx.coroutines.BuildersKt__Builders_commonKt.launch$default(Builders.common.kt:47)
	at kotlinx.coroutines.BuildersKt.launch$default(Unknown Source:1)
	at no.nordicsemi.android.kotlin.ble.server.ServerViewModel$advertise$1.invokeSuspend(ServerViewModel.kt:136)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7924)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
```

reactormonk, 2023-10-25:
I've hooked into the `BleAdvertiserOreo`:

```
$ advertiseData?.toNative()
AdvertiseData [mServiceUuids=[00001523-1212-efde-1523-785feabcd123], mServiceSolicitationUuids=[], mTransportDiscoveryData=[], mManufacturerSpecificData={}, mServiceData={}, mIncludeTxPowerLevel=false, mIncludeDeviceName=true]

$ settings.toNative()
AdvertisingSetParameters [connectable=true, isLegacy=false, isAnonymous=false, includeTxPower=false, primaryPhy=1, secondaryPhy=1, interval=160, txPowerLevel=-7, ownAddressType=-1]
```

when I let the process continue, it'll spit out the above exception.

My device name is "FP3", so that shouldn't push it over the edge.

philips77, 2023-10-25:
Local Name: 5 bytes
Service UUID: 18 bytes

What if you set `isLegacy` to true?

reactormonk, 2023-10-25:
First:
```
java.lang.IllegalStateException: Legacy advertisement can't be connectable and non-scannable
```
Then:
```
java.lang.IllegalArgumentException: Legacy advertising data too big
```

PR #50 (2023-10-31): Implement fixes for issues reported on Github
- Fix #44: Make client device field public in server connection.
- Fix #42: ServerBleGatt.create should have a timeout
- Fix #29: ServerBleGattCharacteristic does not wait for a BluetoothGattServerCallback.onNotificationSent
- Fix #48: Add license
- Fix #31: Initial value of a server characteristic is not set in tests
- Fix #34: Stuck Reads when disconnected
- Fix #33: ApplicationScope's dispatcher can't be changed in tests
- Fix #35: Concurrency issues when reading rssi
- Fix #39: Advertising fails to enable in the Blinky App
- Fix #30: Initial va [... cut]
closed 2023-10-31 by sylwester-zielinski