nordicsemi/Kotlin-BLE-Library #62: Something broke 1.0.6 => 1.0.8 notifying subscriptions as server

reactormonk, 2023-11-03:
I have a server where `setValue` is being called on a regular basis, but when upgrading 1.0.6 to 1.0.8, only first value gets propagated, the client doesn't receive updates anymore.

sylwester-zielinski, 2023-11-06:
New function has been added `setValueAndNotifyClient` which sets value, notify listeners and waits for a proper callback to finish. `setValue` is now used only for setting value for READ operations. I will change the name of this to maybe `setLocalValue` to avoid confusion in the future.

reactormonk, 2023-11-07:
Or go for `setValueForRead`, and `setValueAndNotifyClient`?

philips77, 2023-11-07:
Or `setValue` and `setValueLocally`? I understand the first one sets also for read and notifies all subscribed clients, the other one just sets the value.

Regarding versioning, it should all be 0.x-alpha, i think. Lots of things may still change. Please forgive us :)

PR #76 (2023-11-15): Fix #62: Change confusing function names for setting value on a server
Fix #62: Change confusing function names for setting value on a server
commit 01ef8e13 (2023-11-15): Merge pull request #76 from NordicSemiconductor/fix62/localValue

Fix #62: Change confusing function names for setting value on a server
commit ceaad426 (2023-11-15): Fix #62: Change confusing function names for setting value on a server
closed 2023-11-15 by sylwester-zielinski