nordicsemi/Kotlin-BLE-Library #63: Manufacturer Specific Data is improperly parsed

ryanmitchener, 2023-11-03:
I switched from https://github.com/NordicSemiconductor/Android-BLE-Library to use this today and during scanning, manufacturer specific data is not being parsed properly. I am seeing two ids when there should be one and both DataByteArrays are empty. I can see the proper bytes in the `bytes` property of the scan record. `Android-BLE-Library` parses it correctly along with NRFConnect.

WkGl001, 2023-11-08:
I am also running into this issue. From what I've seen in a debugger, this seems to be happening during the conversion from a ScanRecord to a BleScanRecord in the "ScanRecord.toDomain" function in no.nordicsemi.android.kotlin.ble.scanner.data. The SparseArray mapping in no.nordicsemi.android.common.core:

```
fun <T, R> SparseArray<T>.map(
    modifier: (T) -> R,
): SparseArray<R> {
    val newArray = SparseArray<R>(this.size())
    for (i in 0..this.size()) {
        newArray[i] = modifier(this[i])
    }
    return newArray
} 
```

treats the SparseArray as a dense array in the for loop. I don't expect this to map the data correctly unless the SparseArray keys are exactly inside that [0, this.size()] range. 

I am working with a peripheral that advertises one piece of manufacturer specific data but the conversion also returns a SparseArray with two empty elements.

sylwester-zielinski, 2023-11-13:
Thanks for the detailed analysis. It helped a lot. The fix should be included in a next version - 1.0.9.

closed 2023-11-13 by sylwester-zielinski