nordicsemi/Kotlin-BLE-Library #64: Bonded devices are returned from scans even if they don't match filter

jpd236, 2023-11-05:
In this code:

https://github.com/NordicSemiconductor/Kotlin-BLE-Library/blob/f1e93f77aa12ebd4468a1b0d124afed1d798d180/scanner/src/main/java/no/nordicsemi/android/kotlin/ble/scanner/BleScanner.kt#L99-L100

All bonded devices are returned as scan results in the output Flow, regardless of whether they match the provided filters. This leads to unexpected results like specifying a scan for a precise deviceAddress and having another device entirely returned, requiring additional post-filtering on the results.

Should filters be applied to bonded devices, or at least should this behavior be documented? Thanks!

Klewerro, 2023-11-06:
It's working exactly the same on my side - only filtering scan results working unfortunately 😢
```fun scanForDevices(context: Context): Flow<ThermometerBleDevice> {
        val filters = listOf(BleScanFilter(deviceName = "LYWSD03MMC"))  // Not working
        return BleScanner(context)
            .scan(filters)
            .filter {
                it.device.name == "LYWSD03MMC"  // Working
            }
            .map {
                aggregator.aggregateDevices(it)
                ThermometerBleDevice(
                    it.device.name ?: "",
                    it.device.address
                )
            }
    }

QuincySx, 2023-11-09:
Me too, I want to add a filter switch, do not want to directly get the device that has been bound.

closed 2023-11-14 by sylwester-zielinski

sylwester-zielinski, 2023-11-14:
Name and address filtering will be from now on applied to bonded devices as well. Also a `includeStoredBondedDevices` option has been added to `BleScannerSettings` which may exclude returning those devices at all.

PR #75 (2023-11-14): Fix #64: Bonded devices are not filtered
Fix #64: Bonded devices are returned from scans even if they don't match filter
commit dfc6ae6a (2023-11-14): Fix #64: Bonded devices are returned from scans even if they don't match filter
commit e965e1ea (2023-11-14): Merge pull request #75 from NordicSemiconductor/fix64/filterBonded

Fix #64: Bonded devices are not filtered