nordicsemi/Kotlin-BLE-Library #85: Need Notification Callback

java4ever, 2023-11-22:
Hello, thank you for providing the library. I'd like to ask how to receive a callback indicating a successful subscription when initiating a GATT notification subscription. Currently, it seems that there is no callback to the caller when writing to the NOTIFICATION_DESCRIPTOR.

juliankotrba, 2024-05-08:
@java4ever I was also wondering about this. In the code there exists the suspending method `enableIndicationsOrNotifications` but since it is private and called in the flows `onStart` callback I am not sure if there is a way to wait for setting up indicate/notify. Would be great if could let me know if I am missing something here.

In another issue I saw that @philips77 is currently working on a refactoring of this library. In case we are not missing something, could this also be addressed in the upcoming version?

philips77, 2024-05-08:
My plan is to enable notifications/indications automatically when `notifications(): Flow<..>` is called. It could be closed with an error if they failed to be enabled.