nordicsemi/Kotlin-BLE-Library #96: Dependency 'no.nordicsemi.android.kotlin.ble:client:1.0.11' requires core library desugaring to be enabled

Klewerro, 2023-12-08:
I just updated libraries (`no.nordicsemi.android.kotlin.ble:scanner` and `:client`) to the newest version in the clean project using only to experiment with this library.
After that, I started to getting error message:
```
Execution failed for task ':app:checkDebugAarMetadata'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.CheckAarMetadataWorkAction
   > 8 issues were found when checking AAR metadata:
     
       1.  Dependency 'no.nordicsemi.android.kotlin.ble:client:1.0.11' requires core library desugaring to be enabled
           for :app.
     
           See https://developer.android.com/studio/write/java8-support.html for more
           details.
     
       2.  Dependency 'no.nordicsemi.android.kotlin.ble:scanner:1.0.11' requires core library desugaring to be enabled
           for :app.
     
           See https://developer.android.com/studio/write/java8-support.html for more
           details.
     
       3.  Dependency 'no.nordicsemi.android.kotlin.ble:client-mock:1.0.11' requires core library desugaring to be enabled
           for :app.
     
           See https://developer.android.com/studio/write/java8-support.html for more
           details.
     
       4.  Dependency 'no.nordicsemi.android.kotlin.ble:mock:1.0.11' requires core library desugaring to be enabled
           for :app.
     
           See https://developer.android.com/studio/write/java8-support.html for more
           details.
     
       5.  Dependency 'no.nordicsemi.android.kotlin.ble:client-android:1.0.11' requires core library desugaring to be enabled
           for :app.
     
           See https://developer.android.com/studio/write/java8-support.html for more
           details.
     
       6.  Dependency 'no.nordicsemi.android.kotlin.ble:client-api:1.0.11' requires core library desugaring to be enabled
           for :app.
     
           See https://developer.android.com/studio/write/java8-support.html for more
           details.
     
       7.  Dependency 'no.nordicsemi.android.kotlin.ble:server-api:1.0.11' requires core library desugaring to be enabled
           for :app.
     
           See https://developer.android.com/studio/write/java8-support.html for more
           details.
     
       8.  Dependency 'no.nordicsemi.android.kotlin.ble:core:1.0.11' requires core library desugaring to be enabled
           for :app.
     
           See https://developer.android.com/studio/write/java8-support.html for more
           details.
```

Links in the error description are suggesting issue with selected JAVA version, but I'm using present version of java. 
```
compileOptions {
        sourceCompatibility JavaVersion.VERSION_17
        targetCompatibility JavaVersion.VERSION_17
    }
    kotlinOptions {
        jvmTarget = '17'
    }
```

What's possible reason of the problem?

dononcharles, 2023-12-20:
To avoid this compile issue, i've set this:
https://android-developers.googleblog.com/2023/02/api-desugaring-supporting-android-13-and-java-nio.html

sylwester-zielinski, 2023-12-21:
Hi, desugaring was enabled by default in our plugins. We don't really need it in the library so I removed it. I believe it should solve this problem.

closed 2023-12-21 by sylwester-zielinski