nordicsemi/Kotlin-BLE-Library #106: ClientBleGattCharacteristic.getNotifications will miss some data when enabe notifications

KittenBall, 2024-01-15:
I create a server and use 
```kotlin
val notificationDescriptor =
        readCharacteristic.findDescriptor(BleGattConsts.NOTIFICATION_DESCRIPTOR)!!
notificationDescriptor.value.onEach {
            onNotificationEnabledChanged(it.value.contentEquals(BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE))
        }.launchIn(connection.connectionScope)

//just send message when notification enabled
private fun onNotificationEnabledChanged(on){
   writeJob = if (enabled) {
            logger.i("$prettyDeviceName 启用通知")
            writeJob?.cancel()
            startWrite()
        } else {
            logger.i("$prettyDeviceName 禁用通知")
            writeJob?.cancel()
            null
        }
}
```

and in client

```kotlin
readCharacteristic.getNotifications()
                .onEach { onMessageReceived(it) }
                .launchIn(scope)
```

when getNotifications called, the library will call write notification descriptor
and most of the time, they work well

but sometimes, because the write descriptor callback and characteristic changed callback sequence dislocation, the data before descriptor callback come in will be ignored, because the getNotifications default buffer(0), but if I pass bufferSize as 1, the message before enable notifications will received.

the log.

server
```log
2024-01-15 13:56:13.064 sdb              V  On gatt event: DescriptorWriteRequest(device=RealClientDevice(device=5D:3E:A9:14:0C:A5), requestId=2, descriptor={ uuid : 00002902-0000-1000-8000-00805f9b34fb, permissions : 17, value : (0x) 00:00, }, preparedWrite=false, responseNeeded=true, offset=0, value=(0x) 01:00)
2024-01-15 13:56:13.066 sdb              I  蓝牙设备：未知设备[5D:3E:A9:14:0C:A5] 启用通知
2024-01-15 13:56:13.078 sdb              V  On gatt event: NotificationSent(device=RealClientDevice(device=5D:3E:A9:14:0C:A5), status=GATT_SUCCESS)
2024-01-15 13:56:13.094 sdb              D  write:(0x) 7E:7D:00:00:00:6D:01:78:9C:75:CC:31:0E:83:30:0C:46:E1:BB:FC:73:2A:61:07:C7:71:36:2A:18:58:38:41:16:50:32:74:C8:D2:96:09:71:F7:22:75:66:7E:9F:DE:81:F6:45:22:87:1D:E9:61:CC:DE:2B:77:3E:44:E9:55:25:76:D1:61:43:C2:91:B1:7F:EA:7B:2E:F9:46:FD:FB:B2:B6:7A:89:8C:71:1A:06:36:91:D1:9E:92:71:C2:A1:BD:0A:12:4B:EC:89:D4:84:82:DA:B5:A0:70:FE:00:FB:0A:1F:C8:30:0A
```

client
```log
2024-01-15 13:56:13.236 sdb              D  Enable notifications on characteristic - start, uuid: fe7e82bc-8cc5-4a09-bef7-c3a90497b82e
2024-01-15 13:56:13.240 sdb              D  Write to characteristic - start, uuid: 769ecb07-8cc2-4254-96b2-6ef7c1bf45a0, value: (0x) 7E:7D:00:00:00:09:00:62:43:5F:73:73:34:33:61:30:0A, type: DEFAULT
2024-01-15 13:56:13.326 sdb              D  Write to descriptor - start, uuid: 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01:00
2024-01-15 13:56:13.328 sdb              V  On gatt event: CharacteristicWrite(characteristic= { uuid : 769ecb07-8cc2-4254-96b2-6ef7c1bf45a0, instanceId : 42, permissions : 0, properties : 8, writeType : 2, value : (0x) 7E:7D:00:00:00:09:00:62:43:5F:73:73:34:33:61:30:0A, descriptors : [] }, status=GATT_SUCCESS)
2024-01-15 13:56:13.332 sdb              I  Value written: (0x) 7E:7D:00:00:00:09:00:62:43:5F:73:73:34:33:61:30:0A to 769ecb07-8cc2-4254-96b2-6ef7c1bf45a0
2024-01-15 13:56:13.401 sdb              D  Write to characteristic - start, uuid: 769ecb07-8cc2-4254-96b2-6ef7c1bf45a0, value: (0x) 7E:7D:00:00:00:52:01:78:9C:AB:56:CA:2D:51:B2:32:D4:51:2A:05:93:49:4A:56:4A:D5:31:4A:A5:C5:A9:45:9E:29:31:20:21:08:C7:2F:31:37:15:C8:8D:51:7A:B6:B5:FB:C5:FA:A9:31:4A:B5:4A:3A:4A:B9:99:29:4A:56:46:A6:16:26:86:86:E6:96:86:26:46:66:26:06:16:46:16:16:B5:00:43:CA:19:78:30:0A, type: DEFAULT
2024-01-15 13:56:13.420 sdb              V  On gatt event: CharacteristicChanged(characteristic= { uuid : fe7e82bc-8cc5-4a09-bef7-c3a90497b82e, instanceId : 44, permissions : 0, properties : 18, writeType : 2, value : (0x) 7E:7D:00:00:00:6D:01:78:9C:75:CC:31:0E:83:30:0C:46:E1:BB:FC:73:2A:61:07:C7:71:36:2A:18:58:38:41:16:50:32:74:C8:D2:96:09:71:F7:22:75:66:7E:9F:DE:81:F6:45:22:87:1D:E9:61:CC:DE:2B:77:3E:44:E9:55:25:76:D1:61:43:C2:91:B1:7F:EA:7B:2E:F9:46:FD:FB:B2:B6:7A:89:8C:71:1A:06:36:91:D1:9E:92:71:C2:A1:BD:0A:12:4B:EC:89:D4:84:82:DA:B5:A0:70:FE:00:FB:0A:1F:C8:30:0A, descriptors : [{ uuid : 00002902-0000-1000-8000-00805f9b34fb, permissions : 0, value : (0x) 01:00, }] }, value=(0x) 7E:7D:00:00:00:6D:01:78:9C:75:CC:31:0E:83:30:0C:46:E1:BB:FC:73:2A:61:07:C7:71:36:2A:18:58:38:41:16:50:32:74:C8:D2:96:09:71:F7:22:75:66:7E:9F:DE:81:F6:45:22:87:1D:E9:61:CC:DE:2B:77:3E:44:E9:55:25:76:D1:61:43:C2:91:B1:7F:EA:7B:2E:F9:46:FD:FB:B2:B6:7A:89:8C:71:1A:06:36:91:D1:9E:92:71:C2:A1:BD:0A:12:4B:EC:89:D4:84:82:DA:B5:A0:70:FE:00:FB:0A:1F:C8:30:0A)
2024-01-15 13:56:13.421 sdb              V  On gatt event: DescriptorWrite(descriptor={ uuid : 00002902-0000-1000-8000-00805f9b34fb, permissions : 0, value : (0x) 01:00, }, status=GATT_SUCCESS)
2024-01-15 13:56:13.421 sdb              D  Write to descriptor - end, uuid: 00002902-0000-1000-8000-00805f9b34fb, value: GATT_SUCCESS
```

the time in two devices is not sysnchronized, but the event order is still relevant.

As so far, my workaround is that delay 1sec when notification descriptor on.

Can anyone provide a better solution?

Thank you.

ThibautPerrinDiabeloop, 2024-01-24:
I am using version 1.0.9 (I cannot upgrade more because of the version of material3) and it seems that I have the same kind issue.

philips77, 2024-02-07:
Fixed by #115, right?

closed 2024-02-07 by philips77

KittenBall, 2024-02-07:
Sorry, I have no conditions to test now, because of the Chinese Spring Festival holiday.

But I don't think this fix is fundamentally solving the problem, because calling enableNotifications ahead of time theoretically only reduces the probability of this problem occurring.

I will test again after Chinese new year.

Thank you very much.