nordicsemi/Kotlin-BLE-Library #109: Crash when device connect in onPhyUpdate.

KittenBall, 2024-01-20:
```throwable
java.util.NoSuchElementException: Key RealClientDevice(device=47:B6:13:A5:56:30) is missing in the map.
        at kotlin.collections.MapsKt__MapWithDefaultKt.getOrImplicitDefaultNullable(MapWithDefault.kt:24)
        at kotlin.collections.MapsKt__MapsKt.getValue(Maps.kt:360)
        at no.nordicsemi.android.kotlin.ble.server.main.ServerBleGatt.onPhyUpdate(ServerBleGatt.kt:309)
        at no.nordicsemi.android.kotlin.ble.server.main.ServerBleGatt.access$onPhyUpdate(ServerBleGatt.kt:87)
        at no.nordicsemi.android.kotlin.ble.server.main.ServerBleGatt$1.invokeSuspend(ServerBleGatt.kt:163)
        at no.nordicsemi.android.kotlin.ble.server.main.ServerBleGatt$1.invoke(Unknown Source:8)
        at no.nordicsemi.android.kotlin.ble.server.main.ServerBleGatt$1.invoke(Unknown Source:4)
        at kotlinx.coroutines.flow.FlowKt__TransformKt$onEach$$inlined$unsafeTransform$1$2.emit(Emitters.kt:223)
        at kotlinx.coroutines.flow.SharedFlowImpl.collect$suspendImpl(SharedFlow.kt:382)
        at kotlinx.coroutines.flow.SharedFlowImpl$collect$1.invokeSuspend(Unknown Source:15)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)
        at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:584)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:793)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:697)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:684)
        Suppressed: kotlinx.coroutines.internal.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@fc3bea1, Dispatchers.Default]
```

I'm trying to reproduce it.

MkhytarMkhoian, 2024-02-06:
Could you make new release with this fix

closed 2024-02-07 by philips77

philips77, 2024-02-07:
> Could you make new release with this fix

Released as 1.0.15