nordicsemi/Kotlin-BLE-Library #119: Dependency on Material3

ThibautPerrinDiabeloop, 2024-02-09:
When using this library (`no.nordicsemi.android.kotlin.ble:client`), we get a dependency on `androidx.compose.material3`
When we already use material3 with a different version, versions are conflicting...

Material 3 seems to be used only by `app_client` module. Why this dependency is embedded in `no.nordicsemi.android.kotlin.ble:client` ?

philips77, 2024-02-09:
Good catch. Let me check it.

philips77, 2024-02-09:
OK. I did some investigation. The `:client` and some other modules depend on `libs.nordic.common.core`, which depends on `androidxcompose.material3`. But the only classes used from that module are `DataByteArray`.
I will remove the dependency to `common.core` and will solve the issue some other way.

So what looks like minor dependency issue will bring breaking changes in the API, moving classes between modules and refactoring. I will keep you posted.

philips77, 2024-02-14:
As far as I remember we needed the `ModalBottomSheet` and swipe to dismiss. Now, with 1.2.0 being final I'm changing the dependencies and removing unnecessary dependencies. Should be ready in couple of days, but with breaking changes.
I'm also polishing the API of the library (class names) whenever I find inconsistency.

I'd like also to remove `DataByteArray` (see [here](https://github.com/NordicPlayground/Android-Common-Libraries/pull/85)) and migrate to `ByteArray`. If you need such class for array wrapper (to avoid `equals()` and `hashCode()` this can be added in the apps, not library. Integer extraction will be done using extensions for `ByteArray` (see [here](https://github.com/NordicPlayground/Android-Common-Libraries/pull/83)).

PR #138 (2024-04-23): Dependencies updated
This PR updates dependency versions. It also removes dependency to Nordic Common library, which fixes #119.

Changes:
1. `DataByteArray`  moved from Commons:core to local package,
2. As logging utils were removed from Common library, this PR migrates logging in BLEK to use `slf4j-api`. Logs can be integrated with LogCat using `libs.slf4j.timber` dependency, like shown in _:app-client_ and _app-server_.
3. Log messages were improved and simplified.
closed 2024-04-23 by philips77