nordicsemi/Kotlin-BLE-Library #123: Not able to get notification after a re-connect i.e. autoConnect = true

d199, 2024-02-22:
I connect with the device with the following: 
`BleGattConnectOptions(autoConnect = true, null, closeOnDisconnect = false)`

Everything works fine; however, when the device disconnects with reason.isLinkLoss = true and then re-connects automatically, there are no new notifications : 
 ```
_TxChar.getNotifications()
                .onEach {  Log.i(TAG, ">>>>>>>>>>>>UART data") }
                .catch { it.printStackTrace() }
                .onCompletion {  Log.e(TAG, ">>>>>>>>>>>>UART notification flow completed") }
                .launchIn(scope)
```

In other words, no "UART data" is printed, nor the 'UART notification flow completed' is printed.

Upon re-connect after the  'On connection state changed: STATE_CONNECTED, status: SUCCESS', there is nothing in the log. 

Any ideas, what am I missing ?

d199, 2024-02-26:
Hi Alex,
Yes. I think so. Apparently the SDK enables the notifications as part of the .getNotifications() during a fresh CONNECT but fails to do so in the case of a reconnect. I was able to get the notifications working by externally enabling them.
The older Java SDK had an explicit public call to enableNotifications so it was very clear to a client they need to manage this aspect; however the Kotlin SDK is trying to do this implicitly but only does it half way.

philips77, 2024-02-26:
OK, I get it. I'm also new to this library, took it over from a colleague of mine, so it will take some time before I learn enough to fix such isses.
I'm also working on refactoring it and releasing as 2.0. I want to make `:client-android` and `:client-mock` independent and optional, at least that's the plan. Some classes and methods may be renamed as well.

d199, 2024-02-28:
I contrasted the behavior with the older SDK behavior. The older SDK goes thru a full discovery and notification enablement. So I changed my implementation to do an explicit flow collection (as opposed to launchIn() ) and cache the resulting job. When I get a disconnect, I cancel this job and start fresh with .getNotifications(). This seems to have resolved the issue. Not sure, if this was the original intent behind the design of the API.

closed 2024-02-28 by d199

philips77, 2024-02-28:
Great, thanks for the info!