nordicsemi/Kotlin-BLE-Library #129: ClientBleGatt.connect Intermittently Hangs

WkGl001, 2024-03-14:
Hello, 

I've been using ClientBleGatt.connect(...) to connect to a ServerDevice retrieved from scanning. Once in a while, the connect function will hang. I first ran into this while still using version 1.0.7 of this library. I am currently updating to use the latest version, 1.0.15, but I have seen it happen in this latest version as well. The major difference with this newest version is that the connect call can now be cancelled so it is at least recoverable. 

This issue has been difficult to reproduce in a normal environment. I've only been able to reproduce this semi-reliably by either: 
1. Having the NRF Connect App connect to the bluetooth device that I'm using first and then connecting to it through my own application.
2. Using breakpoints in a debugger when connecting with my own application.

From what I've seen poking around in the library so far, it looks like there may be a race condition after connectGatt is called on the BluetoothDevice object with the gattCallback (ClientBleGattFactory.kt line 167) between when that gattCallback gets a connection state change to propagate into the ClientBleGatt object and when the ClientBleGatt::waitForConnection function sets the onConnectionStateChangedCallback (ClientBleGatt.kt line 183) so that it can be resumed. 
The callback passed in to Android can be called asynchronously. I have seen the connection state change event get triggered and propagated on one thread before the ClientBleGatt's waitForConnection could set its own onConnectionStateChangedCallback on another thread.

Thank you.

philips77, 2024-04-23:
Hi, I'm working on rewriting the library. New version won't be backwards compatible, as there's too many changes.

waqasakrambueid, 2024-05-19:
I am also having this same issue when Connect hangs and never releases. I spent almost a week investigating this and found that someone is a race condition and it is somehow bug internally in the library.