nordicsemi/Kotlin-BLE-Library #133: Documentation Improvement - Device Connect

ecitulsky-ims, 2024-03-22:
As per the readme, to connect to a device the following example is given:

`    //Connect a Bluetooth LE device. This is a suspend function which waits until device is in conncted state.
    val connection = blinkyDevice.connect(context) //blinkyDevice from scanner`

The comment "blinkyDevice from scanner" is confusing. The scanner returns a BleScanResult object that does NOT have a connect function. Where does the blinkyDevice come from? My request is for a better example of scanning -> connect. I searched the entire project for an example of this as well, but could not find anything. The Blinky sample app seems to be built off of a the old nordic library as well.

philips77, 2024-03-25:
Hi,
I'm working on refactoring the project (version 2.0) and will also fix the doc.
For now, to connect to the scanned device use `ClientBleGatt.connect(...)`:
https://github.com/NordicSemiconductor/Kotlin-BLE-Library/blob/11c4883d6d8fbda6ca7deda884930f6adef75317/client/src/main/java/no/nordicsemi/android/kotlin/ble/client/main/callback/ClientBleGatt.kt#L513-L532

philips77, 2024-03-25:
You may find a sample in nRF Toolbox: https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/a253906cf9edbf1ff826e748a730f9c3f42bb602/profile_hrs/src/main/java/no/nordicsemi/android/hrs/service/HRSService.kt#L91

philips77, 2024-07-05:
Version 2 release status -> https://github.com/NordicSemiconductor/Kotlin-BLE-Library/issues/152