nordicsemi/Kotlin-BLE-Library #136: Serious stability issues with Gatt operations upon auto-reconnect

d199, 2024-04-10:
I've been running into a couple of stability issues with the latest release 1.0.15.

The probable pattern is as follows:

1. Client device moves away from the server and gets disconnected, which causes a suspended writeCharateristics to fail
```kotlin
14:07:09.662 17773-17820 BLEK-LOG         V  On gatt event: CharacteristicWrite(characteristic= { uuid : 6e400002-b5a3-f393-e0a9-e50e24dcca9e, instanceId : 13, permissions : 0, properties : 12, writeType : 1, value : (0x) , descriptors : [] }, status=GATT_ERROR)
14:07:09.664 17773-17820 BLEK-LOG         E  Write to characteristic - error, uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e, result: GATT_ERROR
14:07:09.690 17773-17820 BLEK-LOG         V  On gatt event: ConnectionStateChanged(status=UNKNOWN, newState=STATE_DISCONNECTED)
14:07:09.690 17773-17820 BLEK-LOG         D  On connection state changed: STATE_DISCONNECTED, status: UNKNOWN
14:07:09.691 17773-17773 App                  E  ACK error Gatt operation failed with exception: GATT_ERROR 
                                                                GattOperationException(status=GATT_ERROR, cause=java.lang.Exception)

``` 

2. Client device comes in range and re-connects automatically with the server. Since the library does not enable the indications automatically upon a reconnection, application tries to do it but the call hangs. I've observed the same issue, if I try to call discoverServices() as well.

```kotlin
14:34:05.591 17773-17820 BLEK-LOG          D  On connection state changed: STATE_CONNECTED, status: SUCCESS
14:34:05.608 17773-17773 App                   D  Connection State : STATE_CONNECTED, status = SUCCESS in [main]
14:34:05.634 17773-17773 App                    I  UART enableNotification: read descriptor
``` 

3. Any subsequent calls to writeCharateristics hangs as well
```kotlin
14:34:17.836 17773-17773 BLEK-LOG                 D  Split write to characteristic - start, uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 24:52:45:43:26:51:3D:31:0D, type: DEFAULT
14:35:39.794 17773-17773 BLEK-LOG                 D  Split write to characteristic - start, uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 24:52:45:43:26:51:3D:31:0D, type: DEFAULT

``` 
Even if you disconnect, the flow associated with notification never closes properly. You've to kill the app and start again -- very annoying.

Further the library does not communicate any state when auto-reconnecting to a server device and if the server device is outside of range. The Android Java SDK generates a DEVICE_CONNECTING event in such a case.

IMHO this library has serious issues with its suspension handling design.  The older Java library with its callback approach is rock solid and easy to debug!

Any idea what could be going on here ?

Thanks

d199, 2024-04-10:
Any idea about the timeline ? I mean are we talking weeks or months ?
Since we've already invested considerable time in refactoring things for the Kotlin SDK and switching it back to Java one would take some more effort  -- just trying to assess how should we proceed.

philips77, 2024-04-11:
I'd say around one month, I want to finish it as quick as possible.
Usable native part should be ready first, mocking will come shortly after. I'll push my changes as soon as I have something to show.

I'm designing the API now. I'd appreciate early feedback if possible.

d199, 2024-04-11:
Thanks. In that case, I would wait.
If it helps, I'll be more than happy to help test new APIs, on a branch -- whenever you are ready. I just need the core stuff - scan, connection/disconnect, discover, r/w char/descriptors, close.

Let me know, if I can be of any help.

philips77, 2024-04-24:
Yesterday I released the current version of the library with minor dependency updates, mostly related to some cleaning in Nordic Common library. E.g. the `DataByteArray` was removed from there, so I added it in BLEK.

The new version, that I'm working, is still in progress. I'm trying to have native client ready asap and will release something here. I'm doing a lof of refactoring even in my new code to make it the way I want.

d199, 2024-05-13:
Hi Alex,
Just wanted to touch base as to how things are coming up...

philips77, 2024-05-13:
Hello,
Things that I have in place:
1. Scanning
2. Connection
3. Requesting MTU, connection parameters (priority), PHY, RSSI
4. Advertising

Current TODO list:
1. I need to implement a mock implementation of scanner and features in 3.
2. Perhaps some refactoring when I find necessary
3. Service discovery and basic GATT operations

Afterwards I'm planning on releasing version 2.0-alpha01 or at least pushing to GitHub on a Draft PR. I'm still unhappy with my current hierarchy and need to make sure it works with the mock impl before I continue implementing more features.

d199, 2024-06-11:
Hi. Any outlook on the 2.0-alpha01 release ? If you think, its not going to happen, pl. let me know, so we can drop this migration.

philips77, 2024-06-12:
I think end of June or 1st week of July an alpha version should be out.
I had to switch context to a different project with higher priority, but it's getting to an end.

philips77, 2024-07-05:
Version 2 release status -> https://github.com/NordicSemiconductor/Kotlin-BLE-Library/issues/152

d199, 2024-07-12:
Thanks for the updates. Based on your [guidance ](https://github.com/NordicSemiconductor/Kotlin-BLE-Library/issues/152), we've decided not to pursue this library in the near future.
Till Kotlin SDK is in alpha, it would be helpful if you can pl. remove the following warning from the Java SDK

> Warning

> This library is getting replaced by our new [Kotlin BLE Library](https://github.com/NordicSemiconductor/Kotlin-BLE-Library).

Its quite misleading and creates a confusion among our customers.

closed 2024-07-16 by d199