nordicsemi/Kotlin-BLE-Library #139: Missing and outdated documentation

piefel, 2024-04-25:
I was pointed to this library over from the classic [Android-BLE-Library](https://github.com/NordicSemiconductor/Android-BLE-Library/) and was at a loss how to start. Is there no documentation, or am I just unable to find it? The README is quite short, and it is outdated.

To be a bit more concrete:
 - Scanner: What’s the aggregrator for, how to actually use the found devices when we need a bit more than MAC addresses.
 - Client: You cannot `connect()` directly on the device (anymore?), and how do you handle connection changes etc.

philips77, 2024-04-25:
I am rewriting the library at the moment, with the native client coming in few weeks. It won't be compatible with the current version. Until then, I recommend using the old Java-based library, despite what it says.

In the new one I hope to have more self-readable API and better doc as well.
I'll be pushing my code when only it gets usable so you can track it.

abrown-sram, 2024-05-21:
How is the new library coming?

philips77, 2024-05-22:
I aim to have the client ready in June, but at the moment I'm busy with another project so can't promise anything.

philips77, 2024-07-22:
See #152.

closed 2024-07-22 by philips77