nordicsemi/Kotlin-BLE-Library #152: Release Date for Version 2.0.0

ahmadshoh-orc, 2024-06-26:
Hello,

I've read through various issues and noticed mentions of an upcoming version 2.0.0. I am about to start a new project and need to decide whether to wait for the new release or proceed with the current version of the Android-Ble-library with Java. Could you please provide an estimated release date for Version 2.0.0?

Thank you for your help.

philips77, 2024-06-26:
Hi,
I'd recommend the current Android-Ble-library with Java. I hoped to have an alpha version before I go for vacation (end of next week), but some other tasks are popping up like mushrooms after a rain, so it has a high risk.
The Java BLE library is popular and went through lots of fixes, so should be stable. I can't say when the Kotlin BLE 2.0 will reach similar status.

Try to keep BLE API isolated so future migration will be possible :)

philips77, 2024-06-26:
And sorry for the delays! We're doing everything we can to push projects fast.

closed 2024-06-26 by philips77

philips77, 2024-07-05:
Hello,
I pushed my changes to _version/2.0_ branch.

### Supported features:
1. Client
   1. Scanner
   2. Service discovery
   3. Reading / Writing / Subscribing to values
   4. Reading RSSI
   5. Reading / setting preferred PHY
   6. Subscribing to / requesting connection parameters
   7. Refreshing cache (probably, maybe some delay may be required)
   8. Stub implementation for `@Preview`s, etc.
2. Server
   1. _Not stable_ API, but no implementation. 
3. Advertiser

### Not (yet) supported features:
1. Client
   1. Bonding / GATT operations when bonded / subscribing to bond state
   2. Mock client implementation
2. Server
   1. Implementation
   2. Mock implementation

philips77, 2024-07-05:
Scanning sample:
https://github.com/NordicSemiconductor/Kotlin-BLE-Library/blob/264f9b5843a75094eab883a3f2e20f18e18b43ee/sample/src/main/java/no/nordicsemi/kotlin/ble/android/sample/scanner/ScannerViewModel.kt#L75-L112

philips77, 2024-07-05:
Connection sample:
https://github.com/NordicSemiconductor/Kotlin-BLE-Library/blob/264f9b5843a75094eab883a3f2e20f18e18b43ee/sample/src/main/java/no/nordicsemi/kotlin/ble/android/sample/scanner/ScannerViewModel.kt#L115-L225

philips77, 2024-07-05:
How to get an instance of a `CentralManager`?
Here's how:
https://github.com/NordicSemiconductor/Kotlin-BLE-Library/blob/264f9b5843a75094eab883a3f2e20f18e18b43ee/sample/src/main/java/no/nordicsemi/kotlin/ble/android/sample/di/CentralManagerModule.kt#L66

There's also `CentralManager.Factory.mock`:
https://github.com/NordicSemiconductor/Kotlin-BLE-Library/blob/34e04bfbc84e32fe5ae3c1f8a96f5e4ff3b40f2d/client-android-mock/src/main/java/no/nordicsemi/kotlin/ble/client/android/mock/MockCentralManagerEngine.kt#L53-L67
but it's not working yet.
The idea is that using the `environment` one may emulate any platform and possible issues.

juliankotrba, 2025-02-13:
Hi @philips77, really looking forward to try out the version 2 of the library!
Just wanted to ask, have you planned to release the current state on maven central as a _2.0.0-alpha-xxx_ or so?
Or is there an other easy way to get an _aar_ without checking out the project and building it ourselves?
Thanks!

philips77, 2025-02-13:
There is a very easy way to try it yourself.

You need to clone the Kotlin BLE library to the same root folder (e.g. StudioProjects) and just add this 3 lines to _settings.gradle.kts_:
https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/1a1f82f017de8f67fd949769b9d9c8fde5b4b92b/settings.gradle.kts#L75-L77

The rest gradle files you configure normally, using `implementation "no.nordicsemi.kotlin.ble...."` as if you'd use them from Maven Central. Android Studio is smart enough that it matches the group names to the included library.

philips77, 2025-02-13:
Just remember to checkout the _version2_ branch!

juliankotrba, 2025-02-13:
@philips77 nice, thx for the quick answer!

philips77, 2025-02-13:
I'm very intersted in your feedback. You'll be one of the first users except our team.

cullub, 2025-02-21:
Hi @philips77 I'm with a device manufacturer that uses Nordic chips and we're hoping to migrate to a native Android app starting in the next 6 months or so. If version 2.0 will be in a beta state soon and stable state in the next 6-9 months, we'd love to use this rather than implementing the Android-BLE library. Any ideas as far as timeline for this project?

philips77, 2025-02-24:
Client implementation is almost feature complete.

All the "common" features should be already operational:
* scanning with/without filters using Low Latency mode
* connecting / disconnecting
* service discovery
* writing / reading characteristics
* enabling / disabling notifications / indications
* subscribing to notifications / indications as flow
* writing / reading descriptors 
* requesting maximum MTU
* requesting PHY / listening to PHY changes
* requesting "connection priority" / listening to connection param updates as flow
* reading RSSI

Unimplemented features:
* scanning with other modes than Low Latency, including one with `PendingIntent`
* reliable write
* mock implementation of a peripheral

cullub, 2025-02-26:
That's awesome. Once everything is feature complete, it would be great to have a 2.0.0-alpha release available through maven central

philips77, 2025-03-03:
Sure thing!

juliankotrba, 2025-03-05:
Hi @philips77! Just wanted to let you know that I have started integrating the new version of the library. After solving all the dependency and build setup issues/conflicts I am progressing very slowly because there is not much documentation yet. For example, I tried to connect to one of our devices. As far as I have found out I need to create a `Peripheral` for do that. Since we do not scan and connect directly, I checked how to create a Peripheral and saw, that you have to pass an `Executor`, which holds the relevant connection details. The `NativeCentralManagerImpl` just creates an instance of a `NativeExecutor` based on the scan result. However, since the class is `internal`, I am not able to do that in the parent project. Probably I am just missing something but sadly there is currently no time to _"play around"_ more. I will wait for a more advanced state of the library but I really appreciate all your efforts you are putting into this new version. Thanks and keep up the good work!

philips77, 2025-03-05:
Hello,
The API is a bit similar to the one from iOS, which is much better in my opinion that from Android.
On iOS you get a `CBPeripheral` using [`CBCentralManager.retrievePeripherals(withIdentifiers:)`](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/retrieveperipherals(withidentifiers:)).

Ini this library you should also use `CentralManager` instance and this method:
https://github.com/NordicSemiconductor/Kotlin-BLE-Library/blob/89c3f13ddedc322f433e15a5bbf45d11833cabf6/client-core/src/main/java/no/nordicsemi/kotlin/ble/client/CentralManager.kt#L80

Here's a snipet:
```kotlin
val centralManager = CentralManagerFactory.native(context, scope)
val peripheral = centralManager.getPeripheralById(bluetoothDevice.address)
```

Internally, it will call this method:
https://github.com/NordicSemiconductor/Kotlin-BLE-Library/blob/89c3f13ddedc322f433e15a5bbf45d11833cabf6/client-android/src/main/java/no/nordicsemi/kotlin/ble/client/android/internal/NativeCentralManagerImpl.kt#L169-L188

mobilekosmos, 2025-04-14:
"reliable write" ? 🥲

mobilekosmos, 2025-04-15:
What do you actually mean by reliable write Philips77? Because above you also wrote:

"All the "common" features should be already operational:
...
wrote writing / reading characteristics
...
"

philips77, 2025-04-16:
Reliable write is a special type of writing value of a characteristic. It's also known as "queued write".
The client writes values using Prepare Write commands (just like with Long Write). Each value written is replied back with the same value to confirm received data. Then, after writing all data (there may be multiple Prepare Write operations) clients sends Execute or Abort operations. On Execute the written data are passed to the higher layer as atomic operation. On Abort the data are discarded as if nothing happened.

Have a look at [`beginReliableWrite()`](https://developer.android.com/reference/android/bluetooth/BluetoothGatt#beginReliableWrite()) and [`executeReliableWrite()`](https://developer.android.com/reference/android/bluetooth/BluetoothGatt#executeReliableWrite()) in Android API and [Bluetooth Core 5.4 specification, Volume 3, Part G, chapter 4.9.5](https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-54/out/en/host/generic-attribute-profile--gatt-.html#UUID-7fc3dafa-7199-3f9c-a137-1575597b2a8c).

Normal Write and Write Without Response are of course supported.

mobilekosmos, 2025-05-08:
Wouldn't it be possible to get some pre-alpha on maven and updated docs how to use the lib? I'm just writing new ble code using Jetpack's never finished ble lib, but would like to have a second implementation using this one.

philips77, 2025-05-09:
I'll try to release it next week.

In the meentime, you may include the library as modules to your project using this code in `setings.gradle.kts`:
https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/cdb3278bc268d337bc1c569be95c3b76871b92e5/settings.gradle.kts#L76-L78

You need to clone the project and checkout branch `version/2.0`.

Regarding documentation, there is a short one on the Readme file and there's a lot of in-code documentation.
I'll improve it and release on GitHub Pages when I have time to resume work on this project.

mobilekosmos, 2025-05-09:
Thanks, I will wait for the maven version.

philips77, 2025-05-30:
I've tagged the release. I'm now trying to publish it to Maven.

philips77, 2025-05-30:
Released:
```
implementation "no.nordicsemi.kotlin.ble:client-android:2.0.0-alpha01"
implementation "no.nordicsemi.kotlin.ble:client-android-mock:2.0.0-alpha01" // WIP
implementation "no.nordicsemi.kotlin.ble:advertiser-android:2.0.0-alpha01"
implementation "no.nordicsemi.kotlin.ble:advertiser-android-mock:2.0.0-alpha01"
```