nordicsemi/Kotlin-BLE-Library #156: Gatt server connection not cancelled

antonio2904, 2024-07-21:
In the Gatt Server, neither gattServer?.cancelConnection(device) nor gattServer?.stopServer() is cancelling the existing connection.

Is there any other way to cancel the ongoing connection with a client device?

philips77, 2024-07-22:
The only reliable way to cancel a connection is from the device side.  You may want to implement a command, which when sent from the phone, will close the connection.
Otherwise, you'd have to make sure all `BluetoothGatt` objects on the phone open for that device are closed, and they may be in different apps. Calling `gatt.disconnect()` only disconnects one of the client connections, but Android will keep the link open until the last client is disconnected.

antonio2904, 2024-07-22:
This problem is only there in kotlin library. Checked using the Android BLE library and it is working fine and the client is receiving the disconnection properly.