nordicsemi/Kotlin-BLE-Library #162: awaitClose{} not triggering issue when stopAdvertise() is called?

shiyu-sz, 2024-10-04:
I implemented the ble peripheral device function according to the method in ServerViewModel.kt, but I found that it can broadcast after calling advertise(), but calling stopAdvertise() cannot stop broadcasting. I thought it was a problem with my code, and then I ran the "Blinky Server" application, and I found that this application could not stop broadcasting either. After debugging, I found that awaitClose was not called.

[image]

philips77, 2024-10-07:
Hi, could you try the new version of the library from _version/2.0_ branch?
It's not source-compatible with the one on _main_, but I hope this new version has more intuitive API. Client and advertiser should work. The new version isn't available yet on Maven Central, so you'd have to add it from source.

shiyu-sz, 2024-10-10:
The program of version/2.0 branch cannot run normally. When selecting "Advertiser"->"Start Advertiser", no broadcast is found and the interface automatically stops broadcasting. When selecting "Scanner", the application will crash.

philips77, 2024-10-13:
Perhaps you run the "mock" flavor. Change to "native". I'm still working on the mock client.