nordicsemi/Kotlin-BLE-Library #164: CONSULTATION CASE OF USE

Beckmann0o, 2024-10-21:
Hi guys. Good afternoon !
I am facing a requirement in which I must integrate my mobile app with a BLE bracelet, basically I need to ensure that my application can scan BLE devices, looking for a BLE packet pre-configured on the bracelet, if the packet is detected, The application will trigger a panic event.
The doubt I have is the following. When the mobile device locks, that is, the screen turns off, the library has the ability to still continue performing periodic scans, for example every 30 seconds, scan for 10 seconds. I wanted to know if there is a possibility of integrating this liberia with a foreground service on Android. Thank you !!

Greetings from Argentina.

philips77, 2024-10-21:
Hello Argentina!

The best way to scan in background is by using this API:
https://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner#startScan(java.util.List%3Candroid.bluetooth.le.ScanFilter%3E,%20android.bluetooth.le.ScanSettings,%20android.app.PendingIntent)
It is working on Android Oreo and newer. You register a `PendingIntent` to a list of `ScanFilter`s. When a matching device is found the `Intent` will be fired, no matter if your app was killed or not. This is the most "battery friendly" way.
You may also start a _foreground service_ of type [_copnnectedDevice_](https://developer.android.com/develop/background-work/services/fg-service-types#connected-device) (I think) and scan more actively. You may configure the effectiveness of scan using [`setScanMode`](https://developer.android.com/reference/android/bluetooth/le/ScanSettings.Builder#setScanMode(int)) in `ScanSettings.Builder`.

Greetings from Poland!

Beckmann0o, 2024-10-21:
Thank you very much for answering, it is very good to have the support of someone so experienced, thank you very much, I will try to implement the proposed api. Greetings from Argentina.

closed 2024-12-06 by philips77