nordicsemi/Kotlin-BLE-Library #170: MTU Limit Question

obermillerk, 2024-12-09:
My coworker and I have a difference in understanding of the `requestMtu` method's limit values specifically around the Android 14 MTU changes. the limit set [here](https://github.com/NordicSemiconductor/Kotlin-BLE-Library/commit/ea951257e72a7fbdc8ef998c2e373c8b33ebbc2e#diff-8eb49edc3d33100e4878ea607270e77487de07ee6ba017f320e294a7599e392fR140) is 517, which is the new limit in Android 14. My coworker is concerned that this should be 512 instead, noting the discussion in [this issue](https://github.com/NordicSemiconductor/Android-BLE-Library/issues/541) from your other library.

My understanding is that the discussion and associated changes from your other library are discussing limiting payload input size in your API for sending so that it will always fit in the MTU even with the changed behavior near the new limits. However in this library it is just setting the MTU on the Android side of the connection, which can go to 517 safely, and either android will ensure that any packets sent will fit their payload in the MTU or it's the user's responsibility to check the MTU.

My coworker is concerned that this limit should be lowered to the data limit (512), worrying that setting this to 517 would allow data up to 517 bytes per packet.

If you could just clarify for us what the actual function of setting the MTU to 517 is, we'd greatly appreciate it!

philips77, 2024-12-09:
Hi,
Since Android 14, I think, whatever param you will give, will request highest possible MTU, that is 517. 
This will allow you to send 512 bytes of data.

In general, teh Att(ribute layer) MTU contains 2 bytes for attribute handle and one for operation, so 3 bytes are "eaten".
That's why the lowest MTU is 23 which allows sending 20 bytes of data.

However, with this 517 I believe there's something else added here. Some additional 2 bytes, but I don't know what for.
The max size of a characteristic value is 512, so it's not possible to send longer data packets anyway, even with Long Write/Long Read. You may send multiple packets, one by one, to achieve that.

TL;DR
In general, Android requests 517, which allows sending 512 bytes. The peripheral should reply with max supported value.

obermillerk, 2024-12-09:
so calling `requestMTU(517)` will not allow sending more than 512 bytes in a packet, correct?

philips77, 2024-12-09:
Right. It also depends on the peripheral how much it supports. The MTU negotiations settle on the lowest supported value by both sides.

closed 2024-12-09 by obermillerk