nordicsemi/Kotlin-BLE-Library #174: Add suspend `discoverServices()` function

cullub, 2025-02-26:
I like the idea of peripheral having a `services()` flow, but in practice we really can't do anything with our connection until after services are discovered, so it would be nice to have a suspend function that will discover services and wait until they're available before releasing the suspend. It could throw an exception if the device wasn't connected, etc. Thoughts?

philips77, 2025-03-03:
You may use `.first()`, the method from `Flow`. It suspends and returns the first services.
To skip the empty set, you may also add a filter before calling `.first()`:

```kotlin
val services = peripheral.services
    .filter { $0.isNotEmpty() }
    .first()
```
(somethign like that)

philips77, 2025-03-03:
Some perihperals change it's attribute database during the connection. A `Flow` is always returning the _current) list of services. It also changes to empty list when disconnected, preventing user from using them.

cullub, 2025-03-03:
The `.filter{}.first()` solution will work for me, didn't think of that. Thanks. Do you know if there's any guarantee that when one service is discovered, all services are discovered?

philips77, 2025-03-04:
Yes, on Android it will return all services. Acutally, the library may (or not yet) filter the services to those specified in the `services(...)` method.

closed 2025-03-07 by cullub