nordicsemi/Kotlin-BLE-Library #175: V2 documentation/feature: `characteristic.subscribe()` doesn't `setNotifying(true)` automatically

cullub, 2025-02-27:
I hope it's fine that I'm submitting all these issues, I figured since V2 is prerelease you might like some feedback.

In the docs for the `characteristic.subscribe()` suspend function, it says;

> This method suspends until notifications are enabled.

Due to the way bluetooth works, it seems like you meant suspend until we're subscribed to any notifications that are sent, but it would be nice if either the subscribe() function also called `characteristic.setNotifying(true)` to actually enable the notifications (maybe with an optional param to `subscribe()`: `enableNotifications = true`, in case some people want to subscribe but not enable notifications yet). Alternatively, reword the documentation for `subscribe()` so that it clarifies that it only enables the subscription to the notifications, not the notifications themselves.

philips77, 2025-03-03:
> I hope it's fine that I'm submitting all these issues, I figured since V2 is prerelease you might like some feedback

That is more than _fine_ :)

cullub, 2025-03-03:
Ah my issue here must be the same as #176

philips77, 2025-05-30:
I found the issue. 
I mistakenly moved `setNotifying(true)` to `onSubscription` block, which is called when the flow starts being observed.
Instead, as doc says, the `subscribe()` method should suspend until the notifications are enabled.

The fix will be released in alpha02. Thank you for testing and finding this bug! Much appreciated!

PR #184 (2025-05-30): Bugfix: Notifications were enabled too late
This PR fixes #175 and #176.
closed 2025-05-30 by philips77